winder/Universal-G-Code-Sender #2813: 0.01 stepsize behaviour

mikeoverbay, 2025-07-04:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

In my app I set the values of the steps using these API calls.
```
    private static void setStep(JogService j, double v) {
        j.setStepSizeZ(v);
        j.setStepSizeXY(v);
        j.setStepSizeABC(v);
    }
```
When the data is requested by my nano, my plugin collects this data (and more).
`  float stepSize = (float) jogService.getStepSizeXY();`

This will never change if the step in the UI is set to 0.01 for xy.
For whatever reason, It simply ignores updating the current step that I am asking for.
Every part of my app works with any other value but 0.01.. NOPE!
Maybe a debug catch left over?

Thanks for reading :)

mikeoverbay, 2025-07-05:
I found the bug. should be.
jogService.java
```
    private static double divideSize(double size) {
        if (size > 10000) {
            return 10000;
        } else if (size <= 10000 && size > 1000) {
            return 1000;
        } else if (size <= 1000 && size > 100) {
            return 100;
        } else if (size <= 100 && size > 10) {
            return 10;
        } else if (size <= 10 && size > 1) {
            return 1;
        } else if (size <= 1 && size > 0.1) {
            return 0.1;
        } else if (size <= 0.1 && size >= 0.01) {  <--------- 
            return 0.01;
        } else if (size <= 0.01 ) {
            return 0.001;
        }
        return size;
    }
```

breiler, 2025-07-05:
This works fine for when using the menu actions and from the Web Pendant:
https://github.com/user-attachments/assets/12589567-6aad-447e-a325-8cce96815034

mikeoverbay, 2025-07-06:
Thank you proofing this isn't API related.
That's good to know. I have nothing in my code that could filter it.
Im going to have to look in to what api calls the web pendant uses.
I also found out 0.020 does the same thing. It never arrives.

mikeoverbay, 2025-07-06:
This is silly. It is an artifact I can not find where/why/how lol.
I tried every method to change the endian and .01 comes in scrambled.
Guess I'll single step but that got me now where. seems to send it clean.

mikeoverbay, 2025-07-06:
this right here in my code is looking for a string carriage return.
LOL Guess what value a carriage return is.. LOL
The nano thinks the line has been read to the end. LOL It all makes sense now. Thank you Maryjane! 
```
    if (c == '\n') {
      buffer.trim();

      if (buffer.startsWith("SYNC")) {
        lastAckTime = millis();
        waitingForSync = false;

        Serial.print("BUFFER: SYNC [");
        Serial.print(buffer);
        Serial.println("]");
```