winder/Universal-G-Code-Sender #2861: GCode Vizualizer fails to initialize on 64-bit Arch Linux

kevinpowell, 2025-08-11:
### Version

2.1.15

### Hardware / Firmware

GRBL 1.1

### What happened

Running the UGS-platform (linux/64 bit) download on an Arch Linux install (a few weeks since last pacman update).  The app launches and appears to be able to communicate with the GRBL board.  However, upon trying to load a gcode file things don't go well.  The file is not loaded, and the visualizer does not update.  The following error is found in the messages.log:

`SEVERE [com.willwinder.ugs.nbm.visualizer.shared.GcodeRenderer]: Could not initialize gcode renderer
java.lang.NullPointerException: Cannot invoke "eu.mihosoft.vrl.v3d.CSG.rotx(java.lang.Number)" because "back" is null
	at eu.mihosoft.vrl.v3d.CSG.text(CSG.java:2767)
	at eu.mihosoft.vrl.v3d.CSG.text(CSG.java:2750)
	at eu.mihosoft.vrl.v3d.CSG.text(CSG.java:2754)
	at com.willwinder.ugs.nbm.visualizer.renderables.OrientationCube.generateModel(OrientationCube.java:82)`

This does seem to be a *system* dependency problem.  The exact same ugs release works just fine (with the same gcode test file) on Ubuntu 25.04.

### How to reproduce

run UGS-platform on a recent Arch linux.  try to load + visualize any gcode file

breiler, 2025-08-12:
What happens if you download and install UGS from our Linux package on Arch?
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.1.15

We are not responsible for the Arch-package and from my understanding that package is not using the bundled Java version.

kevinpowell, 2025-08-13:
Sorry if my original report was not clear:  This problem happens when running the above linked UGS package -- exactly as downloaded from github releases -- on a more-or-less recent Arch linux install.

For maximum specificity: this is the link to the package that I tried https://github.com/winder/Universal-G-Code-Sender/releases/download/v2.1.15/linux-x64-ugs-platform-app-2.1.15.tar.gz

breiler, 2025-08-17:
I managed to reproduce it with Arch running in VirtualBox. I don't have a solution for it yet.